google/OpenSK #708: Status Update on NFC Support

fsydere, 2024-10-28:
Hello OpenSK team,

I have noticed a few issues and discussions regarding NFC support for OpenSK dating back to 2021. I'm currently exploring implementing NFC functionality with OpenSK, specifically using the nRF52840 platform, and I would like to know the current status of NFC support.

Could you please provide an update on the following points:

Current NFC Capabilities: What is the current state of NFC support in OpenSK? Are there any recent updates or ongoing work in this area (#409 and #315)?

Hardware Requirements: Is there any recommended hardware setup for NFC functionality, particularly concerning external NFC antennas for the nRF52840 Dongle?

Known Limitations: Are there any known limitations or areas that still require development/testing for NFC in OpenSK, especially on power requirements as stated in #15?

Alternative Hardware Platform: Are there any other hardware platforms with NFC support that are compatible with OpenSK? 

Thank you for any information you can provide. Your guidance will be highly appreciated as I continue my work on this implementation.

Best regards,

jmichelp, 2024-10-28:
Hi

NFC is a low priority for us, especially now that all phones come with USB-C.

As you mentioned, we started the work but it's not finished. We happily review code and accept external PRs though :)
Kamran extended the initial NFC support from Mirna to allow fragmented APDUs (which was required for CTAP2 to work) but that was done on a much older version of the codebase and might require some adjustments. We haven't actually tested it since the PR was merged but it should still be part of the CI checks IIRC.
Beside finishing the driver and the capsule, some code refactoring might be needed in OpenSK to allow the packets to come from either USB or NFC and lock on that transport: a request which started on NFC shouldn't be allowed to continue over USB and vice versa.

Power requirements haven't changed because they come from the Nordic chip itself: adding an external NFC antenna to the dongle won't be enough; the chip requires external power (e.g. with a coin cell battery like with devkit board).

As for alternative platforms, we're a small research team and won't have the bandwidth to support more than the platforms we are using to dev/test our code.

fsydere, 2024-10-28:
Thank you so much for your quick response. 😁