google/OpenSK #709: How to disable flash protection to debug second TockOS app (app panic)

janeoa, 2024-11-19:
I am new to OpenSK and TockOS, both feel great so far.
Building the OpenSK for nrf52840DK 131kb ctab2 app that can be loaded into TockOS. The Readme shows an ability to install OpenSK to already existing TockOS installation using `./deploy.py --board=nrf52840dk_opensk --opensk --no-tockos`, which does successfully installs the app and puts it to 0x40000. But the installed app then flashes with all 4 LEDs which indicates "app panic". 

I am using the default TockOS because I have my second TockOS tab app running, but I also heard that the OpenSK's TockOS is patched and now im not sure if the OpenSK is panicking due to unsupported TockOS and I do not know how to debug it yet.

When I install OpenSK without `--no-tockos`, the app works fine, but then `tockloader` does not respond due to flash being locked. Is it possible to deploy OpenSK without blocking the flash and will my second app work on the "patched" TockOS?

janeoa, 2024-11-19:
Generally, does OpenSK's tock even support a second app?

kaczmarczyck, 2024-11-19:
Hi, thanks for the question!
We don't officially support other apps right now, for some reasons:

- The USB driver contains some hacks, and only lets one app connect to it.
- The storage is not secured against access of other apps, and contains all of OpenSKs secrets.
- We are locked on a specific TockOS commit, and [patch it to our needs](https://github.com/google/OpenSK/tree/develop/patches/tock).

That being said, if you want to spend the effort, depending on what your other app does, it might be possible to run them in parallel:

- Your app does not need USB (otherwise, you'd have to rewrite the USB stack to route packets correctly and adapt OpenSK to it, lots of work).
- You fully control and trust the other app so that it doesn't mess with OpenSK storage. If it needs some sort of storage itself, you'd need to change the [layout](https://github.com/google/OpenSK/blob/develop/nrf52840_layout.ld).
- You can port your app to our older patched TockOS version. Or vice versa, but to port OpenSK, you'd also need to rebase the patches.

Also you'd have to check if the binary sizes of both apps together fits the available space. If you don't want to use upgradability, this is only a problem if your other app is big.
When all of that is done, if I didn't forget any major problem, OpenSK can run. There might still be some UX problems though. If your other app uses the buttons or blinks LEDs, there is no good way to tell which app this relates to. You could adapt OpenSK to limit itself to certain LEDs or buttons on the dev board at least. All of this comes down to what the other app is trying to do.

To summarize, there are some unsolved problems around multiple apps. I can maybe give you more concrete ideas if you tell me more about the other app you are trying to run?

janeoa, 2024-11-20:
Thank you for the answer! 

* My app does need USB access and I am currently trying to learn how does TockOS and OpenSK work with it, so I guess ill have to "rewrite the USB stack to route packets"
* I will have "fully control and trust" to my app, but im currently not sure what is "OpenSK storage", so I'll look into it. (thanks for the link)
* I guess my best bet is to use the same "older patched TockOS version" to ensure OpenSK compatibility.

For the upgradability and binary sizes im planning to use customized nrf52 SDK DFU and to have both apps and tockos build into single file and to be verified by the dfu (currently without A/B to ensure it fits).

p.s. Currently OpenSK looks like the best foundation for my app, but im just getting started with nrf52 and tockos, so I'll try my best to understand both :D I'll try to contribute to multi-app USB if my initial attempts are to be productive.

The questions are answered. I'll go do my research.

closed 2024-11-20 by janeoa

kaczmarczyck, 2024-11-20:
For inspiration, there is a hacky option that I only recommend if you just want something working quickly:

You don't write a second app, but integrate your code into the [main loop of OpenSK](https://github.com/google/OpenSK/blob/c5b08fdf481e5d9242a6d1c45bc4d963d1429873/src/main.rs#L182). So whenever you receive a packet, you check if it looks like it's for OpenSK or your other app.

That has some downsides, of course:

- The packets might be hard to distinguish.
- The device doesn't show up with both functions in lsusb.
- You have to entangle yourself with OpenSK's codebase.
- During user presence checks, OpenSK doesn't return to the main loop for some time. While this can be benefitial since the other app can't interfere then, it also blocks for up to 30 seconds.

So only useful for a quick check, not good long term solution.

Please let me know if you make progress with the USB stack (e.g. make OpenSK work with TockOS's own stack)!