google/OpenSK #719: Cannot deploy key using `--vendor-hid`

zumbiepig, 2025-03-05:
Expected behavior:
The security key gets deployed with Vendor HID enabled.

Actual behavior:
```
info: Generating Tock TAB file for application/example ctap2
WARNING:root:Unknown TLV block in TBF header.
WARNING:root:You might want to update tockloader.
info: Generating all-merged HEX file: target/nrf52840_dongle_dfu_merged.hex
info: Creating DFU package
info: Please insert the dongle and switch it to DFU mode by keeping the button pressed while inserting...
info: Press [ENTER] when ready.
info: Flashing device using DFU...
  [####################################]  100%          
Device programmed.
info: Configuring device.
info: Using the Vendor HID interface
info: Programming OpenSK device AAGUID 5f9ab2de-dabd-41af-992c-10a62891f09f (CtapHidDevice('4295196839')).                 
error: Failed to configure OpenSK (unsupported command).                                                                   
100%|████████████████████████████████████████████████████████████████████████████████████████| 1/1 [00:00<00:00, 26.30it/s]
fatal: Could not read device configuration.
```

Steps to reproduce:
```
./deploy.py --opensk --board=nrf52840_dongle_dfu --programmer=nordicdfu --vendor-hid
```

kaczmarczyck, 2025-03-12:
Thanks for reporting this, I can reproduce. Will look into it.

zumbiepig, 2025-03-19:
@kaczmarczyck Any update on this?