google/OpenSK #726: No device found after initial deployment

jojoke, 2025-04-01:
Hi everyone ! 
I hope you are doing well.

I don't quite understand why after initial flashing, my board isn't found by ./tools/configure.py.

My env :
- Host OS : Ubuntu 24.04
- Python 3.9 (a 3.12 is also installed, but forced the default one to be 3.9)
- Board : nrf52840-DK

My command prompt : 
`
git clone -b develop https://github.com/google/OpenSK.git
(skipped all the verbose stuff)
cd OpenSK
./setup.sh
(skipped to keep it readable)
sudo cp rules.d/55-opensk.rules /etc/udev/rules.d/
sudo udevadm control --reload
(disconnect then reconnect the board)

./tools/configure.py \
    --certificate=crypto_data/opensk_cert.pem \
    --private-key=crypto_data/opensk.key
(device not found, but it's normal since it hasn't been flashed yet)

./deploy.py --board=nrf52840dk_opensk --opensk
Identité d'auteur inconnue

*** Veuillez me dire qui vous êtes.

Lancez

  git config --global user.email "Vous@exemple.com"
  git config --global user.name "Votre Nom"

pour régler l'identité par défaut de votre compte.
Éliminez --global pour ne faire les réglages que dans ce dépôt.

fatal: impossible de détecter automatiquement l'adresse ('jojoke@jojoke-Thinkpad.(none)' trouvé)
info: Building Tock OS for board nrf52840dk_opensk
    Finished release [optimized + debuginfo] target(s) in 0.06s
OpenSSL 3.0.13 30 Jan 2024 (Library: OpenSSL 3.0.13 30 Jan 2024)
info: Building OpenSK application
Testing invariants in customization.rs...
   Compiling ctap2 v1.0.0 (/home/jojoke/github/OpenSK)
    Finished `test` profile [unoptimized + debuginfo] target(s) in 9.75s
     Running unittests src/lib.rs (target/debug/deps/ctap2-5b5be3478ea56b46)
   Compiling ctap2 v1.0.0 (/home/jojoke/github/OpenSK)
    Finished `release` profile [optimized] target(s) in 6.17s
info: Generating Tock TAB file for application/example ctap2
info: Erasing all installed applications
info: Flashing padding application
info: Installing Tock application ctap2
WARNING:root:Unknown TLV block in TBF header.
WARNING:root:You might want to update tockloader.
WARNING:root:Unknown TLV block in TBF header.
WARNING:root:You might want to update tockloader.
info: Configuring device.
info: Your device is not yet configured, and lacks some functionality. You can check its configuration status with:

./tools/configure.py

If you run into issues, this command might help:

./tools/configure.py \
    --certificate=crypto_data/opensk_cert.pem \
    --private-key=crypto_data/opensk.key

Please read the Certificate considerations in docs/customization.md to understand the privacy trade-off.
fatal: No device to configure found.`

I understand that, at first ./tools/configure.py shouldn't work, but why after tockos and opensk was flashed, does tools/configure still do not show ?

Thanks in advance :)

jojoke, 2025-04-07:
> Hi, thanks for the report. Is it generally not working, or only as part of the deploy script? What happens if you run the suggested command manually afterwards?
> 
> ```
> ./tools/configure.py
> --certificate=crypto_data/opensk_cert.pem
> --private-key=crypto_data/opensk.key
> ```
I get this : 

```/tools/configure.py     --certificate=crypto_data/opensk_cert.pem     --private-key=crypto_data/opensk.key
info: Private key is valid.
/home/jojoke/github/OpenSK/./tools/configure.py:105: DeprecationWarning: datetime.datetime.utcnow() is deprecated and scheduled for removal in a future version. Use timezone-aware objects to represent datetimes in UTC: datetime.datetime.now(datetime.UTC).
  now = datetime.datetime.utcnow()
/home/jojoke/github/OpenSK/./tools/configure.py:106: CryptographyDeprecationWarning: Properties that return a naïve datetime object have been deprecated. Please switch to not_valid_before_utc.
  if cert.not_valid_before > now:
/home/jojoke/github/OpenSK/./tools/configure.py:108: CryptographyDeprecationWarning: Properties that return a naïve datetime object have been deprecated. Please switch to not_valid_after_utc.
  if cert.not_valid_after <= now:
info: Certificate is valid.
fatal: No devices found.
```

any command using ./tools/configure.py doesn't work.

I checked, and basically fido2.hid.CtapHidDevice.list_devices() returns nothing.
I thought it was normal the first time because the OS wasn't flashed yet (therefore it's not yet a Ctap device).

Unless there is something to flash before deploy.py ?

kaczmarczyck, 2025-04-08:
May I ask you to double check a few things?

First, please make sure that this step completed successfully:

```
sudo cp rules.d/55-opensk.rules /etc/udev/rules.d/
sudo udevadm control --reload
```

There should be a file in `/etc/udev/rules.d/`.

The other reason I have encountered why this step fails is if not both USB cables are connected, or one doesn't transmit data. So please check that what you plugged into "USB Peripheral" [as annotated](https://github.com/google/OpenSK/blob/develop/docs/img/devkit_annotated.jpg) works.

If these fail, can you check whether you see the device here?

```
lsusb -v -d 1915:521f
```

jojoke, 2025-04-21:
Ok,
How to recreate my problem.

1. Install nrfConnect (download the .appImage, chmod +X the file then launch it with the option --no-sandbox as a simple user)
2. Install the tool "Programmer" in nrfConnect.
3. Use it to erase all memory on the board, connected with the J-Tag port (you  can also connect the nrf port, but it will be only useful later)

[image]

In that state, the given tutorial won't work, and the hex won't flash.

To make it work, simply flash manually the .hex.

4. Generate the .hex manually using the following command : 
`./deploy.py --board=nrf52840dk_opensk --opensk --programmer=none `  
Using the nrfConnect Programmer, flash the .hex on the board.

5. Redo the tutorial, and at the end of it, don't forget to reset  the board (or power it off and on again).

The board should work now.

One thing that still doesn't work, and that I don't get why, is the certificates :
Even when I launch the command :
`./tools/configure.py \
    --certificate=crypto_data/opensk_cert.pem \
    --private-key=crypto_data/opensk.key
info: Private key is valid.
/home/jojoke/github/OpenSK/./tools/configure.py:105: DeprecationWarning: datetime.datetime.utcnow() is deprecated and scheduled for removal in a future version. Use timezone-aware objects to represent datetimes in UTC: datetime.datetime.now(datetime.UTC).
  now = datetime.datetime.utcnow()
/home/jojoke/github/OpenSK/./tools/configure.py:106: CryptographyDeprecationWarning: Properties that return a naïve datetime object have been deprecated. Please switch to not_valid_before_utc.
  if cert.not_valid_before > now:
/home/jojoke/github/OpenSK/./tools/configure.py:108: CryptographyDeprecationWarning: Properties that return a naïve datetime object have been deprecated. Please switch to not_valid_after_utc.
  if cert.not_valid_after <= now:
info: Certificate is valid.
info: Programming OpenSK device AAGUID 5dca9976-d8ba-43d9-a00a-bf58d1f12e6e (CtapHidDevice('/dev/hidraw2')).                                                                    
info: Please touch the device to confirm...                                                                                                                                     
info: Certificate: Present                                                                                                                                                      
info: Private Key: Present                                                                                                                                                      
100%|█████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 1/1 [00:04<00:00,  4.45s/it]`
I don't have the OpenSK AAGUID.

[image]

jmichelp, 2025-05-08:
AAGUID is not also sent back. It depends on the parameters you set for the request.
Try to go in the advanced options and change Attestation from None to Direct to see if that happens too.

jojoke, 2025-05-21:
Hi, 
sorry for the delay - it does not work - even with direct attestation.

kaczmarczyck, 2025-06-23:
I tried to reproduce the AAGUID issue, and what @jmichelp said is correct:
In webauthn.io, I also don't see a AAGUID for OpenSK, if I don't use attestation. I assume the browser filters it for privacy reasons. However, if you use user verification and Direct attestation, the AAGUID is shown in webauthn.io as expected.

Which browser are you using? Can you confirm that you get asked for a PIN when you do your AAGUID tests on webauthn.io?

To double check the Linux connection problem, can you confirm that you ran

```
sudo cp rules.d/55-opensk.rules /etc/udev/rules.d/
sudo udevadm control --reload
```

and your rules file is in `/etc/udev/rules.d/`?

For both problem, can you try running a simple Python script? It should print the info, including AAGUID:

```
from fido2.hid import CtapHidDevice, CAPABILITY
from fido2.ctap2 import Ctap2

def main():
  for dev in CtapHidDevice.list_devices():
    print(f"Device {dev.product_name}")

    if dev.capabilities & CAPABILITY.CBOR:
      ctap2 = Ctap2(dev)
      info = ctap2.get_info()
      print(f"DEVICE INFO: {info}")
    else:
      print("Device does not support CBOR")

if __name__ == "__main__":
  main()
```

Make sure to install the `fido2` pip package.