google/OpenSK #739: ./tools/configure.py shows no devices after deploy.py step with nordic nRF52840 dongle (not DK/development board)

alriceeeee, 2025-07-19:
After running this command:  ./deploy.py --board=nrf52840_dongle_dfu --opensk --programmer=nordicdfu 

it says:

(py_virtual_env) alrice@ubuntuu:~/OpenSK$ ./deploy.py --board=nrf52840_dongle_dfu --opensk --programmer=nordicdfu --verbose --debug
Author identity unknown

*** Please tell me who you are.

Run

  git config --global user.email "you@example.com"
  git config --global user.name "Your Name"

to set your account's default identity.
Omit --global to set the identity only in this repository.

fatal: unable to auto-detect email address (got 'alrice@ubuntuu.(none)')
info: Building Tock OS for board nrf52840_dongle_dfu
    Finished release [optimized + debuginfo] target(s) in 0.04s
OpenSSL 3.0.2 15 Mar 2022 (Library: OpenSSL 3.0.2 15 Mar 2022)
info: Building OpenSK application
Testing invariants in customization.rs...
   Compiling ctap2 v1.0.0 (/home/alrice/OpenSK)
    Finished `test` profile [unoptimized + debuginfo] target(s) in 7.03s
     Running unittests src/lib.rs (target/debug/deps/ctap2-fe3664be00e923d1)
   Compiling ctap2 v1.0.0 (/home/alrice/OpenSK)
    Finished `release` profile [optimized] target(s) in 5.71s
info: Generating Tock TAB file for application/example ctap2
WARNING:root:Unknown TLV block in TBF header.
WARNING:root:You might want to update tockloader.
info: Generating all-merged HEX file: target/nrf52840_dongle_dfu_merged.hex
info: Creating DFU package
info: Please insert the dongle and switch it to DFU mode by keeping the button pressed while inserting...
info: Press [ENTER] when ready.

info: Flashing device using DFU...
  [####################################]  100%          
Device programmed.
info: Configuring device.
info: Your device is not yet configured, and lacks some functionality. You can check its configuration status with:

./tools/configure.py

If you run into issues, this command might help:

./tools/configure.py \
    --certificate=crypto_data/opensk_cert.pem \
    --private-key=crypto_data/opensk.key

Please read the Certificate considerations in docs/customization.md to understand the privacy trade-off.
fatal: No device to configure found.

after trying to run ./tools/configure.py (and the suggested command)

i get hit with: 
(py_virtual_env) alrice@ubuntuu:~/OpenSK$ ./tools/configure.py     --certificate=crypto_data/opensk_cert.pem     --private-key=crypto_data/opensk.key
info: Private key is valid.
/home/alrice/OpenSK/./tools/configure.py:106: CryptographyDeprecationWarning: Properties that return a naïve datetime object have been deprecated. Please switch to not_valid_before_utc.
  if cert.not_valid_before > now:
/home/alrice/OpenSK/./tools/configure.py:108: CryptographyDeprecationWarning: Properties that return a naïve datetime object have been deprecated. Please switch to not_valid_after_utc.
  if cert.not_valid_after <= now:
info: Certificate is valid.
fatal: No devices found.

any suggestions? lsusb hangs while the device is plugged in so that doesn't help, and dmesg shows my dongle is in open bootloader dfu mode (so default) yet it flashes green instead of pulsing red.

closed 2025-07-19 by alriceeeee

alriceeeee, 2025-07-23:
> I assume you resolved this yourself? For completeness, your device should be flashed. So you don't need DFU mode anymore. The configure command will fail if you boot into DFU mode again, so replugging and using it normally might do the trick?

https://github.com/google/OpenSK/issues/544

The issue is you guys need to tell users to use --erase_storage for it to work otherwise configure.py always breaks

kaczmarczyck, 2025-07-24:
The wording was indeed not specific enough on the main branch. #742