google/OpenSK #743: Device works on windows but not linux?

alriceeeee, 2025-07-25:
[My issue is pretty much the title, i've been able to use the device perfectly on windows, but on linux i get nothing, it shows up in lsusb just fine, there's no problem with dmesg either, it straight up just doesn't work

Steps i've tried:
- Adding rules.d file
- Adding user to plugdev group and restarting

kaczmarczyck, 2025-07-26:
What did you need to change to make it work on Linux?

From what I read on other repos, the flatpak issue is not OpenSK specific:

- https://github.com/ciderapp/Cider-2/issues/935
- https://github.com/ungoogled-software/ungoogled-chromium/issues/1299

alriceeeee, 2025-07-26:
> What did you need to change to make it work on Linux?
> 
> From what I read on other repos, the flatpak issue is not OpenSK specific:
> 
>     * [[Flatpak] Unable to log into Apple account with security-key authentication ciderapp/Cider-2#935](https://github.com/ciderapp/Cider-2/issues/935)
> 
>     * [[Flatpak] Can't login to websites using U2F Device Yubikey ungoogled-software/ungoogled-chromium#1299](https://github.com/ungoogled-software/ungoogled-chromium/issues/1299)

Literally nothing besides changing to the .deb version of a web browser, i could try getting flatpak to run without sand boxing to see if it lets me use them. I'll try and make a list of what browsers work with .deb files

closed 2025-07-28 by kaczmarczyck